Makeblock-official/mBlock #5: mBlock for Linux

BugBuggi, 2015-06-15:
When will be a Linux package available? 

I supported mBot Project on Kickstarter. But I saw that the software is only available for Mac and Windows, but not for Linux. 
As I only use Ubuntu on all Computers, the mBot robots will be useless for me.

As the Software is Open Source, it should be no Problem to also build a Linux Package!

As Linux is the only Open Source operating System of the three, the Open Source mBlock Software should certainly be available for Linux also!!!

I will be Glad to support you with testing Linux packages.

Best Regards
Buggi

evaletolab, 2015-07-08:
+100 that's exactly the same for me. A bit frustrating for my daughter

rootless4real, 2015-07-08:
We are heavy Linux users too. It would be great to have a Linux Package. Thks.

Makeblock-official, 2015-07-10:
the scratch2.0 is based on adobe air2, so be mblock. But it seems that adobe no longer support linux desktops. Maybe in browser scratch-x could be an option to linux users.

sdmeijer, 2015-07-10:
Scratch 2.0 Offline is also available for Linux, so mBlock should be too.
You only have to use an older version of Adobe Air (version 2.6).

evaletolab, 2015-07-14:
I'm not familiar with Adobe Air but mBlock and Scratch seems different in some regards. For scratch there is only one file Scratch-<version>.air, that is not the case of mBot or I missed something?

A simple tutorial to run mBlock on Linux will be nice.

BugBuggi, 2015-07-15:
> some regards. For scratch there is only one file Scratch-.air, that is not

Yes, this would be great, I now have the two robot kits but cannot do
anything with them because of lacking Linux support.
And this is a mess.

Best regards / Saludos Cordiales / Distinti Saluti / Avec mes meilleures
salutations / 心想事成，万事如意 / Mit freundlichem Gruß

Matthias Burkhardt

Gesendet mit Oneplus One / Cyanogenmod 12

nichtich, 2015-07-19:
mBot is surely usable without Scratch by just using the Arduino IDE and programming C code. However it is difficult to find all related C libraries and firmware hidden somewhere in the sources. It would help to package a repository with C libraries and a brief tutorial only.

nichtich, 2015-07-19:
Arduino IDE 1.6 [supports 3rd party hardware](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x---package_index.json-format-specification) - official adding `package_makeblock.cc_PACKAGENAME_index.json` would simplify use a lot!

marmundo, 2015-07-26:
+1 to the Linux Support. The most of the brazilian schools are using Linux. So, if the idea is help children to learn programming, and the most of the schools can use Linux, mblock should try to support linux.

marmundo, 2015-07-26:
Dear friends,

The scratch provides support for Linux. https://scratch.mit.edu/scratch2download/

Why don't try to give a version or explain to want to develop one how to create a version of the mblock.

You could explain in the wiki how the windows and macOS version were created. What adaptation needed has done to the code to get the versions.

Let the community help too.

jnweiger, 2015-08-02:
I also need Linux support for using mBlock with our STEM classes. I can use the windows version myself, and plan the exercises, but in class, we only use Ubuntu machines.

I agree with @evaletolab: It seems that mBlock software does not use adobe AIR at all. So their lack of Linux Support should not hinder us here.

On the other hand, I have not found the build instructions yet, how Mac and Windows binaries are built. Reviewing these would help us better understand the requirements. I don't think these instructions are intentionally hidden. It's under GPL-2.0, and this license clearly says that build scripts shall be included.

cyrilcc, 2015-09-16:
Linux support is also a must have for me.

I managed to run mBlock under Ubuntu with Wine, but I never made mBlock list any serial port, so it's not possible to communicate with the mbot or orion board using USB wire.

evaletolab, 2015-09-22:
My kids cannot do anything because of lacking Linux support. The lack of support is a mess,

BugBuggi, 2015-09-22:
Totally agree with you, same here! The worst thing is that they are now
announcing iOS support and still don't have Linux support. I will never buy
anything from them again.

>:
>
> My kids cannot do anything because of lacking Linux support. The lack of
support is a mess,

BugBuggi, 2015-11-08:
Just saw that scratch is available for ubuntu in the Software center!
http://packages.ubuntu.com/en/precise-backports/scratch
https://wiki.ubuntuusers.de/Scratch

So why is no package for mbot available for ubuntu ? 
It would suffice to pack all the needed libraries in the package as scratch is available over the software center.

Some comment would be highly appreciated.

Regards
Buggi

herrniemand, 2015-11-16:
So much reply! OMG! TECH SUPPORT IS THAT BEST!

daniel-alomar, 2015-11-16:
Please, provide support for Linux

Regards,

Daniel

cmcknight, 2015-11-16:
Hi Buggi,

To be honest, I suspect the issue is that they have limited programming resources to work on mBlock. Scratch is written in ActionScript and takes proprietary tools to build, so it may simply be a case of aiming at their largest perceived market. I’m wondering (and may investigate when I get some free time) whether or not it would simply be a case of replacing the Arduino executables from the Windows or OS X platforms with ones for Linux.

I sent a pull request that they’ve applied to the Makeblock-Library that corrects an issue involving case-sensitive file names (which is not a problem on Windows, but is for any flavor of Unix). To make the development easier and more open, I’ve suggested that they port their code to Snap4Arduino which will give the same graphical block-based environment but it is based on Node.js and Javascript which should make getting contributions much easier given the popularity of Node.js/Javascript these days. I don’t really know of too many people who want to learn ActionScript and those I know really dislike it. 

Hopefully the Scratch team will consider porting to something like that for the next major version of Scratch.

Regards,

Chuck

cozo, 2015-12-01:
Useless without linux paquage

cmcknight, 2015-12-01:
The biggest issue I see is that Adobe discontinued support for Air on Linux which is what the offline editor is based on. I've made a somewhat pointed comment to the Scratch team about that, but they consider their app to be primarily browser-based, so I don't look for changes there any time soon.

Over the holiday, I'm investigating doing a port to Snap! 4.x for Arduino which is based in HTML/CSS/Javascript to eliminate proprietary tools and have a better shot at running on most platforms. Of course, you should be able to run mBlock via Wine or in a Windows VM (unless you are running El Capitan on OS X). Alternatively, you should be able to use the Makeblock libraries with the Arduino programming environment under Linux. That does have the disadvantage of forgoing the mBlock graphical environment, but what the heck, you or your child will run into the walls/ceilings/floors in Scratch/mBlock anyway. I've already bumped up against it not supporting lists or a lot of other options when using the mBot header block because the code generation to do so isn't there (another thing I'll be looking into with the Snap4Arduino port).

Regards,

Chuck

cozo, 2015-12-17:
Did you try web technologie like blocky : https://developers.google.com/blockly/

Maybe it will help. 

Thank you for your job i really hope a linux paquage, proprietary software are not a good idea for educationnal purpose.

cmcknight, 2015-12-18:
Interesting, I hadn't looked at Blockly in a while. I can do native code generation from either, so I'll look for the path of least resistance (and effort). Thanks for the suggestion!

on4tux, 2015-12-19:
I quickly looked at the mblock software, especially in the ext/libraries directory. There are multiple subdirectories with .s2e files that indicate the communication between mBlock and the hardware is done via cleartext .js and .s2e files. Similar to the A4S approach, this leads me to believe they do serial communication via the .js files and that we should be able to adopt these files so you can use the standard Scratch for linux and load these as experimental extensions? Again, this needs further investigation, but perhaps someone who worked with these extensions before can help here?

vernon-sullivan3, 2015-12-25:
I'm trying to make mblock work in Linux using wine. 

It works fine, but I haven't get serial comms working yet. I'm trying some registry tweaks but my serial ports don't even get listed.

Has anybody tried using mblock with wine?

vicinigit, 2015-12-26:
I'd also like to add my support for Linux compatibility. I bought a mBot for Christmas for my daughter and was shocked that mBlock wasn't supported. Her computer runs Ubuntu so I'm not sure what I'm going to do with it now. Quite disappointed it wasn't clearer on the information. I was going to recommend it to friends but won't be able to until there's Linux support.

Anyone had success with WINE or other methods?

Makova, 2015-12-26:
Hi. Run mbot and Wine Scratch 2, but does not recognize serial port. At the end uninstall everything.

q2dg, 2015-12-28:
As @cmcknight has said, Snap (http://snap.berkeley.edu) and Snap4Arduino (http://s4a.cat/snap) are the solution. I opened long time ago an issue on Scratch github about get rid of its dependency of Adobe AIR (https://github.com/LLK/scratch-flash/issues/760) but nobody understood me.

manuti, 2015-12-31:
GNU/Linux support please!

Makova, 2015-12-31:
:+1:  @manuti

manuti, 2015-12-31:
@Makova ya te he visto pajarillo y me he quedado más tranquilo al saber que hay alguien más tratando de echar a andar esto bajo GNU/Linux !!!

daniel-alomar, 2015-12-31:
No answer from developers, so it seems they think they have not market for
GNU/Linux Users.... grrrr...

johnlepikhin, 2016-01-01:
One more vote for Linux version.

AubinMahe, 2016-01-02:
Another vote for GNU/Linux version.

manuti, 2016-01-02:
And if the version is compatible with Raspbian and Raspberry Pi maybe Makeblock will have a killer pair to boost educational robots and computers.

m-elvex, 2016-01-04:
mBlock for Linux is a must

daniel-alomar, 2016-01-04:
If we don't take any action, they will ignore us. I think saying than we
want a linux port in this mails will not affect in any way the company....
So we must take action.. Any ideas?

manuti, 2016-01-04:
Maybe to launch our proper Kickstarter Campaign?

cozo, 2016-01-04:
Just don t buy it its enough i think

manuti, 2016-01-04:
To late @Julien I bought

q2dg, 2016-01-04:
There's an alternative by BQ: http://diwo.bq.com/product/kit-printbot-evolution/
It's IDE is called Bitbloq. The bad news are that it's online and only for Chrome (due to this: https://github.com/bq/bitbloq-serial-uploader), but it's a step forward in order to resolve this issue...

cmcknight, 2016-01-04:
&nbsp;&nbsp;&nbsp;&nbsp;Makeblock appears to be more into the hardware business than the software business from what I can tell. Most of their development seems to be under Windows, with someone kindly compiling a Mac version. I would suggest that if we need a Linux version, it will likely need to be built by the community. I've suggested that Snap4Arduino might be a good starting point because it should only required porting the firmata for tethered operations.
&nbsp;&nbsp;&nbsp;&nbsp;I would also note that the current version of mBlock does not provide full code generation for all Scratch items (lists come to mind immediately). I would rather see an IDE that fully supports two-way code generation between the graphical environment and the Arduino environment because I believe that will provide the most value. Rather than continue flailing away and complaining about the lack of Linux compatibility, it would be more productive for us to discuss how to make such an IDE happen.

&nbsp;&nbsp;&nbsp;&nbsp;Just my dollar and a half...

rootless4real, 2016-01-04:
I agree. Ideas on how to proceed? Other than establishing a code base on
Github or code.google.com. Which code is the best starting point? What
counts as having achieved Linux support; e.g. WINE or no WINE? It might
even eventually get support from Makeblock.

cmcknight, 2016-01-05:
I would suggest that setting up a discussion that covers what everyone thinks should be there, who can work on developing the code, and setting up a Github/Bitbucket/etc. repo. I'm completely booked until after I teach the class using mBots at the end of February, but I should have time to focus on that sort of project afterwards. A good start would be for someone to make a list of all of the fixed blocks and what they can do (parameters, etc.), brainstorm on the custom block creation, and determine what data types/structures need to be supported. All of these are necessary to identify before we start thinking about code generation (Graphical -> Arduino code and vice versa). It would also be helpful to determine what other Arduino code/libraries needs to be generated/included. That would be my suggested starting point for the project. For the record, I really like what Makeblock is doing with the hardware. It's well thought out, and I'd like to see them continue to be successful.

yosauron, 2016-02-01:
+1 
Support for Linux is a must have!

xeecos, 2016-02-01:
http://www.instructables.com/id/Programming-With-Scratch-X-for-Makeblock-MBot/ ubuntu tested

manuti, 2016-02-01:
@xeecos Oh! Thanks!!! I give it a try this evening!

xeecos, 2016-02-01:
@q2dg I install chromium with flash on my rpi2, it can work in scratch.mit.edu and scratchx.org

m-elvex, 2016-02-01:
It's a great step but I think mBlock goes beyond than ScratchX, giving the possibility to convert Scratch in C++.

xeecos, 2016-02-01:
@m-elvex I think we can transplant mBlock to a chrome app integrated with codebender compiling arduino online.

m-elvex, 2016-02-01:
It looks interesting. Go ahead! :)

q2dg, 2016-02-01:
Maybe Visualino can help...

cmcknight, 2016-02-01:
I’d still vote for looking at Snap4Arduino. mBlock really needs to get away from Actionscript because Flash is dying regardless of what Adobe says to the contrary. Snap is written in Javascript, CSS, and HTML and offers first class objects, methods, and variables. It would make doing some kind of list implementation possible that would enable data structures to be created and used. Plus the Snap4Arduino team is interested in having the conversation.

on4tux, 2016-02-01:
@xeecos Thanks for the tip! I confirm that I got it working on both a windows 7 and ubuntu laptop, although I had to change the group setting on the latter for the HID device as explained. This is a decent temp workaround and proves this can work with Scratch. On the downside, we will require internet (only to load the plugin?) and it is subject to changes or removal without our control. An offline version would indeed be much more interesting.

xz81, 2016-02-11:
+1 for linux support

cassianomac, 2016-02-18:
Forget ActionScript... Snap4Arduino is a perfect code base to evolve mBlock. My vote for Linux support! ;-)

cassianomac, 2016-02-18:
I'm just start using Snap4Arduino, it's so easy... very interesting! I'll start teaching Educational Robotics with this software! Sorry mBlock!

juliogilabert, 2016-03-09:
One more vote for Linux version.

npetit49, 2016-03-29:
hello

install 

https://github.com/Makeblock-official/Makeblock-Libraries

and ardublock

https://github.com/Makeblock-official/Ardublock_Installation_Package_by_Makeblock

I am on xubuntu and it's ok for me !

leleoalmeida, 2016-06-16:
@xeecos how do you connect to the mBot via bluetooth? I tried using the Bluetooth assistant both on my raspi 3 running MATE and on a dell running ubuntu LTS. I **REALLY** need this to work for my school.

Help!

manuti, 2016-06-16:
Yes, you can't be open if you don't provide a GNU/Linux solution.

[34 more replies]

closed 2017-02-05 by bigeyex