Makeblock-official/mBlock #7: How to compile the source code?

sdmeijer, 2015-06-26:
Can you provide me the details on how to compile mBlock?

I need a new version with my Dutch-translation in it.
But I need it very soon (Tuesday next week).

Makeblock-official, 2015-06-26:
take a try of Adobe Flash Builder

sdmeijer, 2015-06-26:
Can you be a bit more specific about the actions I have to do? I'm not familiar with Adobe Flash Builder.

sdmeijer, 2015-06-26:
I'll close it, because I don't need to compile it to test my new translation.
Didn't know that mBlock looks at C:\Users\<username>\Documents\mBlock\locale for the language files. ;-)

sdmeijer, 2015-07-05:
I'm reopening my issue. I don't need it for the translation, but I want to add some devices.
But to be able to do that, I first have to know how to compile mBlock.
Can you please provide me with detailed information on how to compile it.
Preferably with a step-by-step guide (incl. how to install the compiling software and how to compile it).

laiqun, 2015-07-11:
could  I help you ?

sdmeijer, 2015-07-11:
@laiqun If you know how to compile it; yes please.

I'm not familiair with compiling Adobe Air. So I need a step-by-step guide (incl. installation instructions).

cymplecy, 2015-07-27:
I would also like to know how to compile it as well

DNROBOT, 2015-07-29:
DNROBOT Help how to compile mBlock. Can you please provide me with detailed information on how to compile it. Preferably with a step-by-step guide (incl. how to install the compiling software and how to compile it).

huzyfee, 2015-12-24:
hi there! please help me. please tell how to compile the source code and which IDE.

ianr34, 2015-12-25:
Under the scratch license - GNU v2 - makeblock must not only release the source code but ALSO how to compile it (or build it). 
ie the license that they have to follow state - 

"The source code for a work means the preferred form of the work for
making modifications to it. For an executable work, complete source
code means all the source code for all modules it contains, plus any
associated interface definition files, plus the scripts used to
control compilation and installation of the executable. However, as a
special exception, the source code distributed need not include
anything that is normally distributed (in either source or binary
form) with the major components (compiler, kernel, and so on) of the
operating system on which the executable runs, unless that component
itself accompanies the executable."

Which MakeBlock are clearly not comply with... I downloaded there so called source code release and it is clearly missing things and has no instructions on how to build it...

So I suspect that it's time we all start complaining to MIT - as if makeblock aren't floowing the license conditions they can't distribute the software...

huzyfee, 2015-12-25:
I second you @ianr34 . you are right I am also thinking to complain. I tried to look into the files. but couldnot get any hint. 
@Makeblock-official kindly tell us step by step guide to compile and build. or we will be forced to complain.

huzyfee, 2015-12-25:
@sdmeijer I have seen your github repository contributing to mblock. Can you please tell us how did you compile the code?

sdmeijer, 2015-12-25:
@huzyfee I only forked it to be able to translate mBlock. Haven't compiled it yet.

cmcknight, 2015-12-25:
There are instructions on the Scratch Wiki [link][https://github.com/LLK/scratch-flash] that describe how to build Scratch (which mBlock is based on). Interested parties might want to have a look...

jm4r7in, 2016-02-22:
+1 @Makeblock-official

cymplecy, 2016-02-23:
I've started reporting this to Scratch https://twitter.com/cymplecy/status/702114615195516928.
Maybe others would care to join in?
I backed this project on the basis that source would be made available - but without instructions the source is pretty useless

JamesH65, 2016-02-23:
Instructions are not required under the terms of the GPL2 licence. Just the scripts, not how to use them. This should be well known in GPL circles.

cymplecy, 2016-02-23:
So what does
"plus the scripts used to
control compilation and installation of the executable." mean?
If the scripts to do this are in the code supplied - where are they?
If you supply them but don't tell anyone where they are is that not effectively the same as not supplying.

And are you the great JamesH from Raspberry Pi come to haunt me elsewhere or is your name just a co-incidence?

sdmeijer, 2016-02-23:
Hi @cymplecy, check your mail.

JamesH65, 2016-02-23:
The scripts are any 'bash' like scripts used to build the code. They
*should* be supplied with the source. There is NO requirement to give
instructions on how to use those scripts. The scripts might be things like
makefiles, configure scripts, install scripts etc.

cymplecy, 2016-02-23:
Well given the lack of useful help your giving I've got to assume it is the same JamesH.  What's the matter - not enough people to attack/denigrate on the forum?  Just let me back in if your getting bored :)

sdmeijer, 2016-02-23:
@JamesH65 mBlock don't supply the scripts to compile the code.
So @cymplecy is right that they don't comply to the terms of the GPL2 licence.

cymplecy, 2016-02-23:
FTAOD I don't care who's right or wrong - we just would like to compile this ourselves so we can develop it further :)

JamesH65, 2016-02-23:
OK, that's how it is, is it? Insult the people giving you information, and in fact the information I gave you is EXACTLY LEGALLY CORRECT, as I have EXACT experience in EXACTLY the same area. You can ignore it, or insult as much as you like, it does NOT make the information less valid. And you really really should care who is right or wrong, because legally, M-BLOCKS are perfectly entitled to not give information on how to build their code. And, if they don't use scripts to build it, then they are in the clear. If it does require scripts to build, and they are not supplied, then they are in contravention of the GPL.  Note, scripts are not necessary to build, if they have supplied makefiles for example.

ArturoAP, 2016-02-23:
:stuck_out_tongue_closed_eyes: Man this is getting really interesting, by the way can someone help me out in the issue i opened #29 ... please?

senestone, 2016-02-23:
Ok, so at the risk of spoiling a perfectly good flame war, I suspect that Makeblock just follows the build instructions for Scratch 2.0 listed [here](https://github.com/LLK/scratch-flash). I'm not the moderator, but we really do need to be respectful of each other whether or not we agree or disagree.

JamesH65, 2016-02-23:
I've had a quick look at the code - it does seem somewhat deficient in build files, but I am no expert in this area. I suggest using the Scratch instructions since this is a fork. Given the company that made it is Chinese, hasn;t updated in a while, and probably not in the slightest bit interested in whether they are a GPL violation or not, I suspect any answers  from them are unlikely.

senestone, 2016-02-23:
I took a peek at the git commit log as saw:

commit 5016e28614e4a78e6a720231db5a0c5d31df0b48
Author: shao-kai <snjdck@live.com>
Date:   Tue Jan 12 17:13:30 2016 +0800

    upload mBlock v3.2.1 source code

The View Source option from within mBlock points you to the Arduino "firmware" source file for the board (mbot_firmware.ino in my case). I'm not really sure why the repos aren't updated more frequently, but there you have it. :-)

It's important to realize that Makeblock is more of a hardware company than a software company, at least at this time. My past experience with hardware companies is that they have fairly limited software resources, and I suspect this is the case here. I continue to reach out to offer support (and will continue doing so) and at some point I'm confident they will come around.

It is likely also important to realize that you can program directly with the Arduino environment. Given the relatively limited amount of variable memory (2K in the case of the mBot), you really shouldn't need to be too concerned with pointers so learning enough C/C++ to write mBot programs should be relatively painless. :-)

ianr34, 2016-02-23:
I had a laugh reading the posts saying makeblock had complied - has anyone actual managed to build their scratch fork? Sure, the firmware for the mbot has finally been released, and it is easy to build, but mblock??

GPL says 
"The source code for a work means the preferred form of the work for
making modifications to it. For an executable work, complete source
code means all the source code for all modules it contains, plus any
associated interface definition files, plus the scripts used to
control compilation and installation of the executable. "

As far as I can tell, they have NOT included all the source code, there are bits missing all over the place, and if they are compiling/building it all by hand without a script or batch file I'll be pretty surprised - and they haven't supplied that either. And where is the source code for their installation script?

The intent of GPL is so that others can expand or modify the work. Is there one single person out there who has managed to do that with mblock? I mean, doing it with scratch 2 is easy, just follow their instructions. Anyone ever made that work for mblock???

senestone, 2016-02-24:
Hi @ianr34,

I'm looking at getting it to compile now. My first step was to make a formal request to the folks at Makeblock to have their developers zip up their development directory so that I can create a patch to include any build scripts. They develop primarily on Windows, so I am not sure what to expect.

My current approach until I hear back from them is to take the Scratch repo and start adding in the mBlock code to see if I can back into getting mBlock built. I'm hopeful that I'll hear something from in the next few days though and I'm happy to share what I hear.

I'm surprised that no one else has taking the step to formally request their build scripts by sending an email to support@makeblock.cc. I would posit that complaining here is akin to shouting down the wind because I'm not sure how closely they monitor the issues for the Github repo. I'm sure they are doing the best they can and I've always found that patience and respectful, professional communications have gotten me further than other approaches especially when dealing with other people from other cultures.

Best regards,

Chuck

senestone, 2016-02-24:
Minor update: Scratch won't build right now. They have an error in their build scripts.  ;-)

cymplecy, 2016-02-24:
Well said Senestone -sorry for my part

ianr34, 2016-02-24:
where did my post go and why (and who)  deleted it ????

cymplecy, 2016-02-24:
I can still see this one https://github.com/Makeblock-official/mBlock/issues/7#issuecomment-187938011

sdmeijer, 2016-02-24:
Last summer a member of the Scratch team helped me compiling mBlock.
It's not perfect and not everything works, but it's a starting point.

It was my plan to figure out how to compile everything myself and then post this online, but unfortunately I didn't have the time to do this.

If you're familiar with compiling Adobe Air applications on Windows, Linux and Mac, please let me know (info [at] fromScratchEd {dot} nl). I'm willing to share the scripts I have, but I'm not putting them online (because they're not completely working and there's no description with it).

There's one rule: If you succeed in compiling mBlock (even partially) you have to share a how-to with me and put it somewhere online (and put the link in a comment in this ticket).

(Oh and another rule: you get the scripts as-is, without any support from me. I wish I could, but my time is (unfortunately) limited.)

ianr34, 2016-02-25:
@cymplecy 
I did one after that, which makeblock have obviously deleted - so they are at least reading this thread!

senestone, 2016-03-03:
Hi All,

I got a reply from Shao-Kai at Makeblock on how to build mBlock. I'm installing Flash Builder Premium now and will report on whether or not I can get it to build.

Below is a rough translation from Google Translate [link](https://translate.google.com/translate?sl=zh-CN&tl=en&js=y&prev=_t&hl=en&ie=UTF-8&u=http%3A%2F%2Fbbs.makeblock.cc%2Fthread-678-2-1.html&edit-text=&act=url):

1. install flash builder 4.7 premium: 
https://creative.adobe.com/products/download/flash-builder 
This really very slow, very slow 

2. download AIR-SDK 
http://www.adobe.com/devnet/air/air-sdk-download.html 

3. download mblock Source: 
https://github.com/Makeblock-official/mBlock/tree/master/source 

4. Unzip AIRSDK and copied to: C: \ Program Files \ Adobe \ Adobe Flash Builder 4.7 (64 Bit) \ eclipse \ plugins \ com.adobe.flash.compiler_4.7.0.349722 \ AIRSDK 

5. Copy the necessary DLL to C: \ Program Files \ Adobe \ Adobe Flash Builder 4.7 (64 Bit) \ eclipse \ plugins \ com.adobe.flash.compiler_4.7.0.349722 \ AIRSDK \ bin dll.zip 

6. Run Flash builder 4.7 
Create a new ActionScript Project 

7. Import code: 
Use mblock \ source \ src replace the newly created project file src 

8. Select the project's properties: 
Add Library folder: mblock \ source \ libs 
Add an external extension:. Mblock \ source \ libs \ * ane 
 
9. Open Mblock-app.xml file, ensure that the second line of the last version before downloading AIRSDK version number: 

    \<Application xmlns = "http://ns.adobe.com/air/application/18.0" \> 
    You can find specific content from AIRSDK / airsdk in. 

10, Compile implementation project files

Laraib-abid, 2016-04-13:
hi senestone  will u please explain the step 5 what is DLL ???

senestone, 2016-04-14:
Not sure what step 5 is. I think it may have to do with the Air dll.

moozam, 2016-05-20:
Hi is there any success to compile mblock sourc code

moozam, 2016-05-20:
and what is step 5 ? please any one explain

laiqun, 2016-06-12:
I had upload my modified packaged in git@git.oschina.net:laiqun/loolPackaged.git. This project is already packaged ,just import project for the loollool.zip file using flash builder 4.7

laiqun, 2016-06-12:
I had packaged my modified project in git@git.oschina.net:laiqun/loolPackaged.git .Just import loollool.zip using flash builder 4.7 .According to @senestone will u please explain the step 5 to solve the problem in debugging .

Laraib-abid, 2016-06-13:
Hi @moozam  here is the complete documnet for compilation hope this will help you 
[HOW TO COMPILE SOURCE CODE OF MBLOCK USING FLASH BUILDER.docx](https://github.com/Makeblock-official/mBlock/files/311364/HOW.TO.COMPILE.SOURCE.CODE.OF.MBLOCK.USING.FLASH.BUILDER.docx)

moozam, 2016-06-13:
thanks i will see thanks again

lemurowaty, 2016-07-04:
Hello, is there anyone who compiled mblock source code? I did instructions from docx file but I got an error message:

> Process terminated unexpectedly.
> 
> The content cannot be loaded because there was a problem loading an extension: Error: Requested extension cc.makeblock.AIRSerial could not be found.
> 
> 
> Launch command details:  "C:\Program Files\Adobe\Adobe Flash Builder 4.7 (64 Bit)\eclipse\plugins\com.adobe.flash.compiler_4.7.0.349722\AIRSDK\bin\adl.exe" -runtime "C:\Program Files\Adobe\Adobe Flash Builder 4.7 (64 Bit)\eclipse\plugins\com.adobe.flash.compiler_4.7.0.349722\AIRSDK\runtimes\air\win" -profile extendedDesktop "C:\Users\Radek\Adobe Flash Builder 4.7\MBlock\bin-debug\MBlock-app.xml" "C:\Users\Radek\Adobe Flash Builder 4.7\MBlock\bin-debug"

FranoTech, 2016-07-05:
Hi. Perhaps the operating system was upgraded and this affected the operation of the program and the robot microcontroller. To solve the problem I recommend you try some of these alternatives:

1. Compiling the code in the Arduino IDE and upload.
2. Update drives microcontroller (chip CH340G / CH341G?).
3. Reinstall the firmware of the microcontroller.
4. Review updates to your operating system and maybe go back to the previous version.
5. Check that an antivirus or firewall is not affecting the connection to the robot.
6. Update the Java VM.

Good luck!

drew345, 2016-07-08:
Did anyone ever figure out what and where to get the "necessary DLL" from step 5 above? It became step 6 in the doc file (Copy DLL files( avoid a flash builder's bug ) ).
Thanks,
Drew

drew345, 2016-07-09:
I am getting same errors as lemurowaty above.
three necessary extension files are missing.
First cc.makeblock.BluetoothExt could not be found so I commented that line out of the xml file.
Then cc.makeblock.AIRSerial could not be found so I commented that line out of the xml file.
Next cc.makeblock.AirHID could not be found either so I commented that line out of the xml file.
Then the program compiled, but got errors that the BluetoothExt, AIRSerial, and AIRHID classes could not be created so the mBlock opened up as a blank screen.

I searched but dont see BluetoothExt, AIRSerial, or  AirHID as standard file types anywhere, I think maybe they are specific makeBlock files.
Any idea where to get the BluetoothExt, AIRSerial, and AirHID extension files greatly appreciated.

Ahrovan, 2016-07-30:
Process terminated unexpectedly

the content cannot be loaded because there was a problem loading an extension.
req extension cc.makeblock.airhid could not be found 

i installed ADOBEAIRINSTALL 21 AND AIRSDK_COMPILER 21
RUN WITH FLASH BUILDER 4.7
 replay plz ...

Tejanshu, 2016-08-07:
I am an absolute beginner at modding Scratch 2.0 derivatives and I am as hassled by the issues in compilation as most of us here. Thanks for valuable inputs in the issue-threads.

Since mBlock is a scratch 2.0 derivative, there is an alternative way of compiling scratch 2.0 source, mentioned here

https://notehub.org/myleo

it looks painful, but it is a different approach from what has been "on record" here and I am inclined to try it.

Whether I succeed and comeback here with to brag about it ... or  I die trying .. I cannot say !

moozam, 2016-10-31:
hi i have compile the code but here is few problem 
"There Are packaging error and warning"

moozam, 2016-10-31:
hi i have following errors

arsh09, 2017-01-27:
drew345, 
can u post the link of the source code of version 3.4.5. All I see is the code for older version with many compilation errors. and I don't know why sometimes ANE works but sometimes AIRHID ane is not found in the same project. :( If I try to export the release build, it builds an executable, but after installation it shows that 'Executable is not a valid win32 app'

closed 2017-02-05 by bigeyex

Nida147, 2018-12-10:
Can anyone elaborate  step#7 and step#8.
i am facing issues with these steps.

7: Import code:
Use mblock \ source \ src replace the newly created project file src

8: Select the project's properties:
Add Library folder: mblock \ source \ libs
Add an external extension:. Mblock \ source \ libs \ * ane

huzyfee, 2018-12-10:
Nida can you please tell me where did you get dll files from in step 5

Nida147, 2018-12-10:
4 links are mentioned below. i have gotten dll files from there.
https://www.dllme.com/dll/files/msvc_dll.html

http://www.dlldownloader.com/qt-dll/

https://www.dll-files.com/cygwin1.dll.html

https://www.dllme.com/dll/files/bluetooth_dll.html

Nida147, 2018-12-10:
> Hello, is there anyone who compiled mblock source code? I did instructions from docx file but I got an error message:

Have you solved this error??

moozam, 2018-12-11:
i have compiled mblock code last year . i remember the issue is in source
code for compiling code we have comment out few lines in source code but
not sure now

Nida147, 2018-12-11:
> i have compiled mblock code last year . i remember the issue is in source code for compiling code we have comment out few lines in source code but not sure now
> […](#)

can you share please?

[1 more reply]