Makeblock-official/mBlock #15: How do you make extensions?

GlennGeenen, 2015-09-17:
Is there any documentation how to create extensions for mBlock? I couldn't find information on it. The extension manager expects a zip file, what should be in this zip file? 

I have scratch extensions I want to add to mBlock so that you can control the mBot with a leap motion or kinect controller. I see in the source files that the mBlock extensions are based on the json syntax used in scratch extensions. What are the required fields, required files?

piif, 2015-10-18:
I've got the same question : I tried to adapt a simple scratch extension i wrote, but it doesn't work.
The only result i can get is reporter blocks which return empty values.
It seems also that mBlock wants something connected even if my extension don't need to communicate with external devices.

Some basic examples should help greatly

ignacioandcorral, 2016-05-08:
yes the same here

closed 2017-02-06 by bigeyex

The-Best-Codes, 2023-03-24:
To make an mBlock extension, use:
https://ext.mblock.cc/