Makeblock-official/mBlock #26: AIRSerial.ane is not the latest version！

V-KING, 2016-01-22:
After I compile the mBlock source，it produce a   "META-INF" directory, then I check the "AIRSerial.dll" file, I find that the size of "AIRSerial.dll" is different. 

The AIRSerial.ane can not work.!

V-KING, 2016-01-22:
oh, and the md5 of the AIRSerial.dll is different.

laiqun, 2016-06-12:
I had packaged my modified project in git@git.oschina.net:laiqun/loolPackaged.git .Just import loollool.zip using flash builder 4.7 .According to @senestone will u please explain the step 5 to solve the problem in debugging . Also fix the AIRSerial.ane error!

codychanghong, 2016-12-30:
hello, did you how to solve this issue? I also meet this problem now, I hope you can help me.Looking for you replying!
Follows is my error log:
Error: Error #2002: Operation attempted on invalid socket.

Fri Dec 30 10:00:05 GMT+0800 2016 : registed:mBot disconnected

启动命令详细信息: "D:\FlashBuilder\Adobe Flash Builder 4.7\eclipse\plugins\com.adobe.flash.compiler_4.7.0.349722\AIRSDK\bin\adl.exe" -runtime "D:\FlashBuilder\Adobe Flash Builder 4.7\eclipse\plugins\com.adobe.flash.compiler_4.7.0.349722\AIRSDK\runtimes\air\win" -profile extendedDesktop -extdir "C:\Users\lenovo\Adobe Flash Builder 4.7\workspace\.metadata\.plugins\com.adobe.flexbuilder.project.ui\ANEFiles\MakerPandaBlock\win32\" "C:\Users\lenovo\Adobe Flash Builder 4.7\workspace\MakerPandaBlock\bin-debug\MakerPandaBlock-app.xml" "C:\Users\lenovo\Adobe Flash Builder 4.7\workspace\MakerPandaBlock\bin-debug"

closed 2017-02-06 by bigeyex

13760209346, 2017-06-27:
Hello, I have solved this problem, and all the processes have been ok, and can be compiled and run, if there are problems you can contact me 13760209346, or add my micro letter or qq

laiqun, 2017-11-06:
easy，just find the ane in the release verson,replace the ane in this repo.
sloved in two year ago.

Nida147, 2018-12-10:
> Hello, I have solved this problem, and all the processes have been ok, and can be compiled and run, if there are problems you can contact me 13760209346, or add my micro letter or qq

what is the solution of this .ane file problem?

roboger, 2019-03-27:
AIRSerial_mac.ane 有没有windos版本的 ，没苹果机器