Makeblock-official/mBlock #29: Spanish translation

ArturoAP, 2016-02-22:
I would also like to help with translations, in this case is with Spanish language since there are still some blocks that are in default (English), but i have a doubt about what file should i update.

In /source/src/locale there are files that correspoond to different languages, but there is also a file named locale.xlx that has all languages in it, what file should i update for the proper translations?.

senestone, 2016-02-23:
I'd suggest updating both the es_ES.po file and the the locale.xlsx file. The .po file is used for the source code, and I suspect the Excel spreadsheet (.xlsx) is just used for reference.

ArturoAP, 2016-02-24:
Ok i will, just another question, i can see another repository called mBlock-languages, has that one been deprecated? or is it also being used for translation?

cmcknight, 2016-02-24:
Can't answer that one. You'd need to contact support@airlockbcc.

laiqun, 2016-06-12:
They also transfer the *.po to locale.xlsx,

senestone, 2016-06-12:
Er, support@makeblock.cc (I am really tired of mobile device autocorrect…)

closed 2017-02-06 by bigeyex