Makeblock-official/mBlock #30: Uploading is unclear - needs to use Arduino mode for ultrasonic to work?

MathBunny, 2016-03-03:
When using an ultrasonic sensor, the firmware is only updated fully when Arduino mode is used to upload the code. If you use the normal upload (through COM) then the ultrasonic won't get any reading after you turn it on/off.

Is it just me experiencing this?

ArturoAP, 2016-04-14:
You mean trough the option "Upgrade Firmware" or "Reset Default Program"?

I will mess around with this options tomorrow using mbots and me orion, i will let you know if i can recreate the issue.

drew345, 2016-07-08:
I see this is old, but having a similar issue.
When I use the ultrasonic on digital pins, it works fine for both upload (Arduino) mode and Scratch (Serial connection) mode. When I use the ultrasonic on the analog pins, it works in the upload mode, but not in the scratch mode. I am trying analog pins A4, and A5. I suspected that the serial mode is using extra power (the US sensor takes a lot of power), but even failed with external power supply. Maybe the serial mode is using those analog pins A4 and A5 for something else and that's why US sensor doesn't work there in Scratch mode.
Anyway, I gave up on it for now, but just saw this very similar post so thought I would add my experience.

closed 2017-02-06 by bigeyex