Makeblock-official/mBlock #32: Release Version vs Source Code Version

ArturoAP, 2016-04-14:
I have a huge question regarding the release version & the source code version, this Monday (4/14/2016) i was surprised that there was a new version of mBlock (yay!), version 3.3.0 to be precise, i went immediately to mblock.cc to check the Release Notes, but it hasn't been updated its still in version 3.2.2, so i came here to check that too, and here there are also no Release Notes, Realizing that there was no explicit release note or changelog, i went into the branches section to check for changes but the only branch available is master and it hasn't been updated in 3 months, What is going on? is the real source code being kept apart from the one here on Github?, am i missing something?

Achifaifa, 2016-07-05:
From what I've seen, apparently all they're doing is just dumping the source code here once in a while.

arsh09, 2017-01-27:
Arturo AP, They have open-sourced the older version (unstable) and have not released the code for the newer version. I also tried to compile the older version which has a lot of errors (somehow I solved a few) but I don't get why the native extensions does not work properly. I added them where they asked in the 'how to compile MBlock' version. I also went to length to de-compile the .swf file of newer version and tweet its code with the older version. Though I am still facing a 'exe not a valid win32 application' error after exporting release build and I have no idea why this error is occurring. Any idea?

closed 2017-02-06 by bigeyex