Makeblock-official/mBlock #37: Problem when upload to arduino card

jpazerty, 2016-06-28:
We work in school, behind samba domain, users are not admin and their profils are on network, when we want to upload to arduino we had this error : 
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
#include <Arduino.h>
#include <Wire.h>
#include <SoftwareSerial.h>

double angle_rad = PI/180.0;
double angle_deg = 180.0/PI;

void setup(){
    pinMode(5,OUTPUT);
}

void loop(){
    digitalWrite(5,1);
    delay(1000*0.2);
    digitalWrite(5,0);
    delay(1000*0.2);
}
C:\Program Files\mBlock\Arduino\arduino_debug.exe --upload --board arduino:avr:uno --port COM3 --verbose --preserve-temp-files U:\config_eole\Application Data\cc.mblock.3.3.0\Local Store\scratchTemp\project____a___a_5_2\project____a___a_5_2.ino
Loading configuration...
java.nio.file.AccessDeniedException: U:\config_eole\Application Data\Arduino15 -> C:\Users\admin.profil\AppData\Local\Arduino15 <=== users have no access to this folder !!!!!!
	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:83)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsFileCopy.move(WindowsFileCopy.java:387)
	at sun.nio.fs.WindowsFileSystemProvider.move(WindowsFileSystemProvider.java:287)
	at java.nio.file.Files.move(Files.java:1395)
	at processing.app.windows.Platform.fixSettingsLocation(Platform.java:219)
	at processing.app.PreferencesData.init(PreferencesData.java:36)
	at processing.app.BaseNoGui.initParameters(BaseNoGui.java:993)
	at processing.app.Base.guardedMain(Base.java:159)
	at processing.app.Base.main(Base.java:137)
send encode mode
 mode binaire   mode charactère
recv encode mode
 mode binaire   mode charactère
	
if we are administrator, all work fine.
Sorry for my bad english.
thank

cmcknight, 2016-06-28:
You'll need to install mBlock in the students' home directories if you cannot change the privileges for the mBlock install. Normally, you could go into the Arduino IDE and just change the path but you may not be able to do that if you have only a single install. At least that's the first thing that comes to mind.

jpazerty, 2016-06-29:
I found a solution 
- create a folder "portable" in mblock/arduino program files directory.
- give access to write for all users in this folder.
and i had a problem too with my users profile, bad path for %localappdata%.
Now all work fine.
Many thanks for your answer

bigeyex, 2017-02-06:
This is also resolved in the newer versions.

closed 2017-02-06 by bigeyex