Makeblock-official/mBlock #41: How to help translating ?

SebCanet, 2016-08-31:
HI,
I made French translation, but I would to correct some sentences and add new ones, how can I do ?

Thanks.

intoeetive, 2016-11-20:
@SebCanet you need to fork this repo, make changes and then make pull request

SebCanet, 2016-11-20:
ok, thank you, Ill do it.

SebCanet, 2016-11-20:
I don't think...it seems there's no more activity on this depot.
The translation way has changed and I see nowhere the file 'locale.xlsx' needed....

intoeetive, 2016-11-21:
It's here: https://github.com/Makeblock-official/mBlock/blob/master/source/src/locale/locale.xlsx

SebCanet, 2016-11-21:
But it's 10 months old...for version 3.2.1. Do they still use it ?

intoeetive, 2016-11-21:
No idea, sir =)

They claim it to be open source, but looking at messages on forums, I can say it's probably not.

SebCanet, 2016-11-21:
And I have no more mail answer ! Dammit...

cmcknight, 2016-11-21:
You might try going on the forums and posting the question to @tec_support.

bigeyex, 2017-01-23:
This is the way: http://www.mblock.cc/posts/note-for-translators

closed 2017-01-23 by bigeyex