Makeblock-official/mBlock #46: compiling source code 3.4.5

drew345, 2017-01-25:
The mBlock 3.4.5 source code has been released now (currently as s branch off the main one). Thought I would start this thread about how to compile and run the code! I got the source code compiled with no errors, but it cannot connect to the mBot or Arduino on the serial port. Under “Connect / Serial Port“, it is just blank, no serial ports show up. The Arduino IDE can see the boards, and the released mBlock 3.4.5 software (from mBlock website) can see the boards. Only the compiled source code of mBlock here cannot see the serial ports.

I compiled the source code with the instructions here:
HOW TO COMPILE SOURCE CODE OF MBLOCK USING FLASH BUILDER.docx

And I exported the completed project as a desktop APP (including Adobe AIR). No errors, and it ran so well, I thought everything was fine! Until I found that the compiled mBlock cannot see any board connected as a serial port.  

I examined the compiled-exported mBlock with a released version downloaded directly from mBlock and found the following folders were missing from the compiled-exported version:
- /arduino directory
- /drivers ch341 directory
- /usbserial directory

I put those three directories into the mBlock folder of the complied-exported version and ran the program again, but still no luck. The compiled-exported mBlock version cannot see the boards on the serial ports.

Can anyone replicate or get past this step??

Thanks,
Andrew

arsh09, 2017-01-27:
Hello Andrew, 
I am not sure where did you get the code for 3.4.5. I downloaded the code for the older version (I guess it is 3.2.1) and then I noticed that it has a lot of bugs like the was Arduino folder when I clicked the 'Arduino IDE' option So I put it there.

About Serial Port, I have noticed that there are three extensions Serial, Bluetooth and AirHID. As mentioned in the -app.xml file as follow 

<extensionID>cc.makeblock.AIRSerial</extensionID>
<extensionID>cc.makeblock.BluetoothExt</extensionID>		
<extensionID>cc.makeblock.AirHID</extensionID>
 but it gives the error that FB can not load the Air HID extension, though when I change the 'cc' in airHID with 'com'. The code compiles in Debug mode and but the Bluetooth option in Connect is grayed out and no serial port is shown. 

If I release the build, after installation, the windows throws 'exe is not a valid win32 error'.

I will be really glad if you could tell where you get the source code of 3.4.5? 

Thank you

arsh09, 2017-01-27:
Andrew, 
I did compile the code with 13 warnings. And I can see the com-port after compilation. I am on Windows so I used the three ANEs ending with 'win'. But I can not export the build. I don't know what is going wrong in it. It builds the export but I am unable to run the mBlock Application. It gives me this error.. Can you help? Are you using the self-created certificate?

[image]

drew345, 2017-01-27:
Arsh.  The 3.4.5 code is a fork off the main branch here.  I made my own
certificate.  I compile with 13 errors just like you.  I unpacked the. exe
file from the mBlock website installer and see that there are 3 drivers
that have to be installed when the code is installed.  I don't have laptop
now but will post that soon.  Anyway,  if you can,  I recommend getting the
3.4.5 installer from the mBlock site. Install it.  Compare installed files
to those created by the source code.  Then also unpack the installer to see
what else is run.  That's where I am at on this.  My contact is
rawjmail1@gmail.com.  Glad to have a partner in this.  It's Lunar new year
so even if we don't hear from mBlock be understanding.  Great to have the
code.  We can do it.  Andrew

arsh09, 2017-01-27:
Andrew
I just noticed on another laptop that you would need to install the drivers to get the com-port. The option is inside the Connect drop down menu. Though you will need to put the Arduino files into the bin-debug folder (or better into the src folder, which will be copied to bin-debug). 

My issue is that I can release the export build but I can not run the application after its installation (or if I use the captive runtime option during the build process then form the MBlock app). 

I am getting the same error i.e. MBlock.exe is not a valid win32 application. Mind that it is the 'Mblock.exe' which I built in the Flash builder. Though the .exe from the MBlock websites works just fine. 

I hope that your serial com-port issue will be solved after you install the drivers.

arsh09, 2017-01-27:
The 13 warnings are about the variable names. I just checked that for example 'len' is used twice in the script 'Face Detection'. I meant it was declared twice. This gives the warning. One can choose to ignore them and can rectify them by changing the name of one of the two variables (but almost everywhere where it is used). 

Funny thing is I can build a blank ActionScript project and can run (or extract and run) its executable files, but when I do the same with MBlock code, windows throws 'exe is not a win32 valid error'.

drew345, 2017-01-27:
I know these images have errors but I think they might help.  It will be a
couple of days until I get back to my computer with flash builder.  I
forgot exactly how I exported but I got a package almost exactly like the
released version and it ran fine except no USB connection

arsh09, 2017-01-27:
Andrew, can you tell me which AIR SDK version u were using when you build? 

I noticed that MBlock always use AIRSDK19 version where I am using AIRSDK21 version. Maybe (hopefully, finger crossed) this may be the reason that the exe is not working fine.

drew345, 2017-01-27:
When you do the export release there was a button to select to include
AIRSDK inside the release.  Or something like AIRSDK.  Be sure to select
that option

drew345, 2017-01-27:
I used  version  24. I thought I might have to use old versions but,, 24
worked fine

arsh09, 2017-01-27:
I am not sure if I can see any such button :(
I go to Project --> Export Build Release --> Select 'Signed Application with Captive Runtime' (or also tried 'native Installer' Option),
Select the build files path and to the Next button, I selected the self-created certificate and its password. and then Finish. 

Did I miss some step in between? 
Sorry for any trouble

drew345, 2017-01-27:
I think your Project - Export is the same I did. The (with Captive Runtime) is what I was remembering.
I don't have my desktop with FlashBuilder with me for 2 more days.

When I added the ANE files, I think I selected:
AIRSerial.ane   (not windows option)  
BluetoothManager.ane     (not windows option)
HID2.4G_win.ane     (here windows option, because only option available)

No Trouble at all! Let's do this. Just sorry I don't have computer with flashbuilder with me now. I am in Korea and it is Lunar New Year holiday. write me and we can talk on skype maybe later. rawjmail1@gmail.com
Andrew

drew345, 2017-01-27:
Ooops, closed this by accident and just reopened.

arsh09, 2017-01-28:
Thanks Andrew. 
This is really weird. I have tried this on three different machines/laptops. I can build release other apps (or blank project) but i cant seem to get pass this error after building mblock.

goddess1992, 2019-07-16:
i have an issue with bluetooth.
every thing is OK and works with no problem but the only problem is that when the robot bluetooth is disconnected software doesn't  recognize it.and when trying to connect robot to software via bluetooth it doesnt connect and we need to restart software .can any body help me please.