Makeblock-official/mBlock #47: locale is not fetched

koenr, 2017-02-09:
First of all: thanks a lot for this Linux port! First tests look great!

When starting mblock, it always starts in Chinese. 
The app.getLocale() function is used, so I expect the language to be pulled from the OS (my locale is nl).
I tried changing the path in linux-unpacked/resources/web/js/libs/translator.js and I tried changing the default locale there, but it doesn't change the behaviour.
It seems to me the path to the locales in that file is wrong (localePath = __dirname + '/../../../i18n/locales) - there is no i18n folder.
I'm not familiar with Electron, so I might have it completely wrong.

paulcoiffier, 2017-02-09:
Open "app/translator.js". On line 74 you have the language hardcoded.
`this.setLanguage("zh_CN");`
Delete this line and replace by another if you want

tomash970, 2017-02-10:
Hi. Tested with one mbot and one computer for now. It works great. Program asked for restart and after that serial port conected to the mbot. Codded just a leds for now.
We have non profit association and 10 ubuntu powered computors for kids to learn coding. Thank you for your effort it means a lot to us. 
I have same issue like koenr, but I can not find "app/translator.js". There is one file named translator.js but it has only 34 lines.

paulcoiffier, 2017-02-10:
Update your repo, I had clone the git repo yesterday so "pull" your git repo

koenr, 2017-02-12:
May be you only looked at the compiled code. It is in the source code. Compiling recipe on http://forum.makeblock.com/t/running-mblock-in-linux/6216/10?u=koenr

bigeyex, 2017-02-27:
fixed in the recent release.

closed 2017-02-27 by bigeyex

koenr, 2017-03-05:
Thanks! Great to see this project moving on!