Makeblock-official/mBlock #49: Doesn't work in linux ia32

joseantoniosa, 2017-02-14:
After several try-and-error Makeblock-4.0.0pre  in amd64,  compiles and works nicely.
Also, Makeblock-4.0.0 pre in ia32 (Inter Centrino Duo) compiles but when I type npm  start   starting window appears and "Loading" text, but nothing else happens.
(nodejs from git repository, I'm downloading Makeblock git )
 How could I debug Makeblock?, Is ia32 architecture supported?
Thank you.

Winux-de, 2017-02-19:
Try to change the libpepflashplayer.so to the 32-bit version.

joseantoniosa, 2017-02-20:
Hello, Me pc is 32bits only. I'll Checa it. Thank you

joseantoniosa, 2017-02-21:
I have tried to install Flash player version 13.12 Firefox and 24.0.0.2.21
Google Chrome.  mBlock git does not show any menú. Any hint to enable
debugging?
Thank you.
José A.

Winux-de, 2017-02-21:
You must change the file libpepflashplayer.so in your mblock/plugins directory.

Search for another libpepflashplayer.so in your system and copy it into the plugin directory or download https://fpdownload.adobe.com/pub/flashplayer/pdc/24.0.0.221/flash_player_ppapi_linux.i386.tar.gz and extract libpepflashplayer.so into the plugin directory

joseantoniosa, 2017-02-21:
Thank you!. Now it works.

closed 2017-02-21 by joseantoniosa

alucasr, 2017-02-22:
It works on my laptop (Ubuntu) but it doesn't work on my son's netbook (Eee PC Asus 1011CX). It has an Lubuntu for i386.
I downloaded the file, unziped and replaced the libpepflashplayer.so on ./resources/plugins/ with the version you mention from adobe.
Do I have to do anything else?
Kind regards
... and thank you very much for this Linux version. Now he can program the Mbot with my laptop... (and hopefully I will get my laptop back when it work on his netbook...)

joseantoniosa, 2017-02-22:
I had similar problems with Ubuntu Precise, it does not work if libc6 version is below 2.17. Type
  dpkg -l libc6
Could you upgrade to another Lubuntu version?

alucasr, 2017-02-23:
In my laptop I have
||/ Nombre           Versión       Arquitectura  Descripción
+++-================-=============-=============-
ii  libc6:amd64      2.23-0ubuntu5 amd64         GNU C Library: Shared libraries
ii  libc6:i386       2.23-0ubuntu5 i386          GNU C Library: Shared libraries

In my son's netbook, it has only the second line (just i386 version and instead ubuntu5 it just says ubuntu, but it's on the same version (2.23). 
ii  libc6:i386       2.23-0ubuntu i386          GNU C Library: Shared libraries

It's a completely updated Lubuntu 16.04 (reciently installed. I have installed nothing on it (neither java, nor Python, ... nothing but system updates))

joseantoniosa, 2017-02-23:
Hello,
 I  have "compiled" a 32 bits version and is it available at [https://drive.google.com/open?id=0Bz4PsRC_QmKENkVUMlRvbEJZbTg]. 
Also you can visit my blog where I have written step-by-step instructions:  [http://tecnologias-secundaria.blogspot.com.es/2017/02/robotica-mblock-en-linux-32-bits.html] (sorry, it s in Galician).
I hope it could help you.