Makeblock-official/mBlock #56: USB DONGLE - Cannot connect to the 2.4G device

profesooooor, 2017-03-02:
When trying "Connect -> 2.4G Serial -> Connect" mblock 4.0 responds with the message "Cannot connect to the 2.4G device. Please check your USB connection or use another USB port".

I have tryied changing the USB dongle from one port to another.
I changed the permissions of every /dev/hidraw* device to crwxrxwrxw.

Unfortunately, all my efforts were useless.

When I launched it from the shell as "pedbel" I could copy this messages:
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
pedbel@INF-2-64bit:/tmp/mBlock$ ./mblock
/tmp/mBlock/resources/tools/arduino
app listening on port 7070!
NOT SANDBOXED
Vector smash protection is enabled.
ready
es_ES:es:en
已更新菜单
已更新菜单
已更新菜单
现在进行2.4G连接。。。
已更新菜单
已更新菜单
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

kiarn, 2017-03-12:
Same here.  Usb connextion works as well.

profesooooor, 2017-03-13:
One more thing.
I usually work with students whose user names are in an LDAP service, inside a server. To make it easier for them to use the mbots without having root permissions, I added a file to the /etc/udev/rules.d directory named "90-mbot-permission.rules" with this contents:
# /etc/udev/rules.d/90-mbot-permission.rules
# Pedro Felip.
# Permitir que cualquier usuario pueda utilizar los mBot
# con cualquier conexión: USB, Bluetooth o 2.4G
# (para poder utilizar ScratchX o Wine/mBlock)
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", MODE="0666"
KERNEL=="rfcomm*", MODE="0666"
KERNEL=="ttyUSB*", MODE="0666"

As a result, every mbot connected with the USB Dongle is apt to be used by any user in the system, no matter his user groups.
I guess it could be possible to make the mblock Linux program in such a way that a user could work with the mbots without being a sudoer.
It could be necessary to install somehing (like the 90-mbot-permission.rules file above) by a superuser. But I cann't allow my students to become superusers. No way.

gebara99, 2017-03-21:
I have the same issue.
But if I run the program as an administrator I don't get the error when connecting the robot (it seems it connects it without problem). The error comes when I execute the program and is a little bit different:

The 2.4G cannot connect the machine.

With the USB I haven't got problems, even with my students with no permission ldap accounts. Everithing works fine.

gebara99, 2017-03-21:
The chinese or japanese letters only say: the menu has been updated so no information in this way

profesooooor, 2017-04-11:
SOLUTION + SUGGESTION

SOLUTION
I have finally found the error.
I am working with LliureX, which is a derivation from Edubuntu (trusty), and mBlock/resources/tools/enableHID.sh has a CR/LF problem, resulting in an error when you try to execute it. To make things worse, it creates the file /etc/udev/rules.? (question mark included), so I did “mv /etc/udev/rules.? /etc/udev/rules.d/10-mblockhid.rules" manually.

I have rewritten the enableHID.sh with a text editor with the same contents and the problem is gone.

SUGGESTION

The file "/etc/udev/rules.d/10-mblockhid.rules" gives permission to every user (666 mode, idVendor 0416, idProduct ffff), no matter his user groups. My suggestion is to comment the line "usermod -a -G plugdev `whoami`" because you don't need to belong to the plugdev user group to make use of the USB Dongle.
Given that, there is no need to populate that user group.

PS. Than you for you nice work!!

gebara99, 2017-04-11:
Hello,
I keep facing the problem.
I have done the new file enableHID.sh and I have checked that I have /etc/udev/rules.d/10-mblockhid.rules. I have this lines there:

```
ATTRS{idProduct}=="ffff", ATTRS{idVendor}=="0416", MODE="666", GROUP="plugdev"
SYSFS{idProduct}=="ffff", SYSFS{idVendor}=="0416", MODE="666", GROUP="plugdev"
```
Is there anithing wrong??

Thank you for your help.