Makeblock-official/mBlock #59: Linux version: Bluetooth problems

emnik, 2017-03-13:
Hi! I am using the 4.0 Linux version. While the bluetooth works if I connect first with blueman (serial port /dev/rfcomm0) and then choose Connect > Serial Port > /dev/rfcomm0 from mBlock, if from mBlock I choose Connect > Bluetooth > MakeBlock(00:0D:19:0E:16:B5) then it connects ok but many blocks from the Robots section (lightsensor on board, ultrasonic sensor Port distance, button on board) does not work. This is rather confusing for everyday. Why one would have to use the serial port menu instead of the bluetooth menu in mblock?... It would be better at least to move the /dev/rfcomm0 option from the Serial Port to Bluetooth Port menu or maybe in both...?

bigeyex, 2017-03-14:
that should be a bug. You do not need blueman or any other software to connect with mBlock. Technically, if you connect with blueman, the bluetooth will be mapped into a serial port.

kajasie, 2017-04-09:
I have a blue-tooth problem too. I'm using Linux 16.04 (kubuntu) with mblock 4.0.0. This is what program says after selecting Connect:
EIO_Init
已找到蓝牙:Makeblock(00:1B:10:10:33:06)
已更新菜单
已更新菜单
已更新菜单
EIO_Init
BTSerialPortBinding::New
EIO_Connect
init from process 3809
init socket: 46 pointer:0x20683178c318 
connect with status: 52
EIO_AfterConnect
BTSerialPortBinding::Close
closing socket: 46
close from process 3809
socket closed pointer: 0x20683178c318
closed with Errorcode 0
bluetooth child process open connect is error:
Error: Cannot connect
    at Error (native)

It seems the robot is connected for a while and then gets disconnected.

skippa, 2017-05-18:
I also have problem with light sensor when connected via Bluetooth, serial is 100% working. What can I do to help with this bug? Can I upload debug logs of some sort?

solrac-ceam, 2017-06-01:
I have the same problem, when I choose Connect > Bluetooth > MakeBlock(00:0D:19:03:6C:E7) all the sensors seem like do not send any value.