Makeblock-official/mBlock #63: Functions are deactivated in mBlock 3.4.6

otmezger, 2017-04-14:
Hi, 
I was wondering why some functions are not active, like for example the events (Q2) or the "When on board pressed" (Q1). See screenshot. What config am I missing? 

Thanks
[image]

bigeyex, 2017-04-14:
Hi, 
When you quit Arduino mode, this block will become available.
This is because the Arduino compiling method mBlock use does not support multi threading, 
so header blocks can be only used in Scratch mode.

otmezger, 2017-04-14:
ok, thanks. not even the "When on board button pressed"? 
[image]

bigeyex, 2017-04-14:
The way mBlock works is converting blocks into Arduino code. The PC is a powerful device, with a large screen and great computing power; the Arduino chip in the robot is not. 

Thus those blocks that make the panda rotate, or run events in parallel, can be only run on PC, not the Arduino-based robot. Even "when button pressed" is a kind of event blocks. Please use repeat and "button ... pressed" instead.

Makeblock is working on a more powerful robot that may support event system - but it is not the entry-level mBot.

closed 2017-04-14 by bigeyex