Makeblock-official/mBlock #65: Mblock not working under Ubuntu 16.10

valentinarolando, 2017-04-27:
Good morning everyone. 
I would like to use mblock to program an Arduino UNO. The final goal is to use Olimex motors, but my problem started way before, I am not capable of uploading with a simple blink example.

I am trying to run makeblock (linux version) with Ubuntu:
Distributor ID: Ubuntu
Description: Ubuntu 16.10
Release: 16.10
Codename: yakkety.

I use KDE as windows manager.

I followed the instructions: downloaded the tar.gz, uncompressed it, checked that the file is executable, launched it.
These are the results:
1 - When I try to run makeblock with my "user account" the windows pops up, but remains stuck during the loading process. The grey wheel keeps spinning (not a graphic server problem though).
The permissions for the folder and the files are correclty set, and my user is already part of the dialout group.
2 - Try again, with super-user this time. The program correctly launches and everything seems fair, I can connect the Arduino, the serial port is recognized and so on, but ... None of the command is actually executed by the Arduino. The red dot in the "robot" section turns grren and on the board the RX led blinks (I suppose something is actually happening), but the Arduino is unresponsive to makeblock commands. 
At this point I tried Conncet > upgrade firmware, but all i get is "Upload failed"
While makeblock is trying to upgrade the firmware, the shell gives me back this:
串口：/dev/ttyACM0
call serial isOpen()
upgrade firmware
已更新菜单
已更新菜单
已更新菜单

Can anyone help please?
Thank you in advance
Valentina

gzuliani, 2017-05-03:
Hi,
I could not connect to the mBot through the /dev/ttyUSB0 port until I added my user to the "dialout" user group:

    sudo usermod -a -G dialout MYUSERNAME

gzuliani, 2017-05-03:
Ops.
I forgot to mention that I'm on an Ubuntu 14.04 box and I'm using the 4.0.0 mBlock release.

Best regards,
Gianfranco

valentinarolando, 2017-05-04:
My user is already part of the dialout group :(
Thank you anyway