Makeblock-official/mBlock #67: mblock can't save the file (mblock 不能貯存檔案)

hkchan, 2017-05-18:
our students finish their mblock homework but sometimes they cannot "save/save as" the file.
Actually, the "save/save as" function sometimes is not working.

student finish their mblock design, and then click "save/save as", select destination path (desktop / D:\ / USB...etc), enter the file name and then click save button, but no any mblock file created.

How can i fix it~

hkchan, 2017-05-18:
Windows 10
mblock version is v3.4.8

bigeyex, 2017-05-18:
This is strange... it works on my machine. What's the file name you are trying to save? It should not be empty.

And would you please try one of the [previous versions](http://www.mblock.cc/download) and reply if that works or not?

hkchan, 2017-05-18:
file name such as "abc" or "12345" is also not working

hkchan, 2017-05-18:
please look at my video: https://youtu.be/gf8yBTeDFTU