Makeblock-official/mBlock #68: Create extension with “header blocks”

Ahmedfcis, 2017-05-20:
I creating an extension contains the following “header blocks”

`[ "h",
  "When Remote %d.IRKeys Key Pressed",
  "procDef",
  "Power",
  {
	"setup":"addPressHandlerIRremote({0},onPress{0});\n",
	"inc":"",
	"def":"",
	"work":"",
	"loop":""
  }
]`

I was expecting to have my generated code be as following:

onPressPower() {
  //...
}

void setup(){
  addPressHandlerIRremote(POWER,onPressPower);
}

but no code was generated, I not sure how 'ArduinoManager.as' handles  “header blocks”

bigeyex, 2017-05-24:
Header blocks are not supported in the current version of mBlock. It will be ignored by the code generator. only "runArduino" header blocks are considered as the start blocks. Implementing this requires an overhaul of the extension system, which is not included in the current batch of iterations.

closed 2017-05-24 by bigeyex

Ahmedfcis, 2017-05-28:
Thank you @bigeyex