Makeblock-official/mBlock #73: mBlock does install but wil not start

jeromba6, 2017-07-29:
I download and install https://mblockdev.blob.core.chinacloudapi.cn/mblock-src/mBlock.deb on my Ubuntu 16.04.2 LTS. The installation did finish, but when trying to start I just get the Panda for a short time blinking in the Launcher and than it disappears. No error's as far as I can see.

What can I do to help fixing this or get is started? 
Is there some logging (can't find it in /var/log/*) ?

AlexeyVAV, 2017-10-04:
I met the same issue.

grregd, 2017-10-14:
Hi

I had the same issue.

Try to run mBlock from command line, then you will see some messages.
I had message like:

`./mblock: error while loading shared libraries`

then the name of the library binary file followed.

In my case the problem was that I have 64 bit Debian, while mblock is 32 bit.
I just had to install few libraries in 32 bit version and it worked.

You just have to figure out the name of the package containing the library.
For example, if it failed to load file `libnss3.so`, you need to install `libnss3`.
To install this library you just execute from terminal:

`sudo apt-get install libnss3`

If you need to install version for particular arch (32 bit for example), you need 
to add appropriate suffix, then command looks like:

`sudo apt-get install libnss3:i386`

Hope this help.

adminies-MD, 2023-03-02:
What about ubuntu 22.04 ?
When i installed this version and libraries libgconf-2-4 or libnss3.
The error message is : app listening on port 7070!  segment violation (core generated)