Makeblock-official/mBlock #75: node  6.11.2 下编译错误

yaoteng, 2017-08-25:
npm ERR! mBlock@4.0.0 rebuild-serialport: `electron-rebuild -f -w serialport`
npm ERR! Exit status 4294967295
npm ERR!
npm ERR! Failed at the mBlock@4.0.0 rebuild-serialport script 'electron-rebuild
-f -w serialport'.

orgrimd, 2017-08-25:
run (for windows) npm install --global --production windows-build-tools
npm install
and then ----> npm run rebuild-serialport

yaoteng, 2017-08-26:
thanks ! but not work

yaoteng, 2017-08-26:
Rebuild Failed

unhandled error occurred inside electron-rebuild
 WARN install got an error, rolling back install
 ERR! configure error
 ERR! stack Error: connect ETIMEDOUT 52.216.192.24:443
 ERR! stack     at Object.exports._errnoException (util.js:1026:11)
 ERR! stack     at exports._exceptionWithHostPort (util.js:1049:20)
 ERR! stack     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1081:1

 ERR! System Windows_NT 6.1.7601
 ERR! command "D:\\Program Files\\nodejs\\node.exe" "D:\\Documents\\Downloads
Block-master\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1
13" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from
urce"
 ERR! cwd D:\Documents\Downloads\mBlock-master\node_modules\node-hid
 ERR! node -v v6.5.0
 ERR! node-gyp -v v3.6.2
 ERR! not ok

led with exit code: 1

or: gyp WARN install got an error, rolling back install
 ERR! configure error
 ERR! stack Error: connect ETIMEDOUT 52.216.192.24:443
 ERR! stack     at Object.exports._errnoException (util.js:1026:11)
 ERR! stack     at exports._exceptionWithHostPort (util.js:1049:20)
 ERR! stack     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1081:1

 ERR! System Windows_NT 6.1.7601
 ERR! command "D:\\Program Files\\nodejs\\node.exe" "D:\\Documents\\Downloads
Block-master\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1
13" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from
urce"
 ERR! cwd D:\Documents\Downloads\mBlock-master\node_modules\node-hid
 ERR! node -v v6.5.0
 ERR! node-gyp -v v3.6.2
 ERR! not ok

led with exit code: 1
 at SafeSubscriber._error (D:\Documents\Downloads\mBlock-master\node_modules\
wn-rx\lib\src\index.js:267:84)
 at SafeSubscriber.__tryOrUnsub (D:\Documents\Downloads\mBlock-master\node_mo
es\rxjs\Subscriber.js:238:16)
 at SafeSubscriber.error (D:\Documents\Downloads\mBlock-master\node_modules\r
\Subscriber.js:197:26)
 at Subscriber._error (D:\Documents\Downloads\mBlock-master\node_modules\rxjs
bscriber.js:128:26)
 at Subscriber.error (D:\Documents\Downloads\mBlock-master\node_modules\rxjs\
scriber.js:102:18)
 at MapSubscriber.Subscriber._error (D:\Documents\Downloads\mBlock-master\nod
odules\rxjs\Subscriber.js:128:26)
 at MapSubscriber.Subscriber.error (D:\Documents\Downloads\mBlock-master\node
dules\rxjs\Subscriber.js:102:18)
 at SafeSubscriber._next (D:\Documents\Downloads\mBlock-master\node_modules\s
n-rx\lib\src\index.js:241:65)
 at SafeSubscriber.__tryOrUnsub (D:\Documents\Downloads\mBlock-master\node_mo
es\rxjs\Subscriber.js:238:16)
 at SafeSubscriber.next (D:\Documents\Downloads\mBlock-master\node_modules\rx
Subscriber.js:185:22)
 ERR! code ELIFECYCLE
 ERR! errno 4294967295
 ERR! mBlock@4.0.0 rebuild-serialport: `electron-rebuild -f -w serialport`
 ERR! Exit status 4294967295
 ERR!
 ERR! Failed at the mBlock@4.0.0 rebuild-serialport script.
 ERR! This is probably not a problem with npm. There is likely additional log
g output above.

 ERR! A complete log of this run can be found in:
 ERR!     C:\Users\Administrator\AppData\Roaming\npm-cache\_logs\2017-08-26T0
2_29_524Z-debug.log

orgrimd, 2017-08-27:
:(( you can remove node-modules and try again. Sometimes windows machines run non-stable :(