Makeblock-official/mBlock #87: mBlock 3.4.5 - Mac OS built

arsh09, 2019-08-18:
Hello, 

I would like to request to developers to kindly release the ANEs for Mac with the build description of x86_64 in ANE extension.xml (as described here, https://blogs.adobe.com/flashplayer/2015/12/air-64-bit-on-mac-osx.html#sthash.w6pDh3kl.dpbs).

The reason why MBlock 3.4 can not be build on MacOS is as follow (according to my research): 

With the release of AirSDK 20.0, you have to specify the platform specifics (x86, x86-64) for MacOS. However, the three important ANEs, (AIRSerial, AirHID and Bluetooth) were compiled using only MacOS x86 options (I confirmed this by installing MBlock 3,4 on Mac OS and looking at the META-INF folder. 

Now I can use AIRSDK 19.0 with Flash Builder 4.7 but there I have the problem that Adobe has limited its support from AIRSDK 20.0 (believe me I have tried all the versions of AIRSDK from 19.0 to 32). 

So I kindly request the developers to either release the code for ANEs (and probably a small guide on how to compile them as ANE) or if they do not want to release the code, please compile the code with MacOS-x86_64 option in extension.xml file of each ANE and release that. 

Thank you 

Arsh

arsh09, 2019-09-30:
Hello, 

If anyone was successful in compiling v3.4.5 for MacOS, please share which AIRSDK was used?

harishree7, 2020-12-23:
> Hello,
>
> I would like to request to developers to kindly release the ANEs for Mac with the build description of x86_64 in ANE extension.xml (as described here, [...]

Have you get ANEs for MAC? If yes, can you provide?

arsh09, 2020-12-23:
https://github.com/Makeblock-official/mBlock/tree/master/libs

Thsee ANEs work. 

Also, I did not use Flash Builder to build this. You can use Eclipse or some other IDE with proper build tools to build for MacOS.

closed 2020-12-23 by arsh09

harishree7, 2020-12-24:
> https://github.com/Makeblock-official/mBlock/tree/master/libs
> 
> Thsee ANEs work.
> 
> Also, I did not use Flash Builder to build this. You can use Eclipse or some other IDE with proper build tools to build for MacOS.

i have used intellij. but it showing this error.

[AIR Debug Launcher]: The content cannot be loaded because there was a problem loading an extension: Error: Requested extension cc.makeblock.AIRSerial is not supported for MacOS-x86-64.