Makeblock-official/mBlock #88: make a fork of mBlock and name question

enteropositivo, 2019-10-31:
I'm thinking in making a fork of mblock 3.4.5

If i have success compiling it, could i name my new forked app : **mblock_MOD** ?  ( as a modified version of this mblock version)

I ask this, because mblock is a registered trademark but mblock is also OpenSource

bpau, 2019-10-31:
The latest mBlock3 from Makeblock to me is 3.4.12 .
mBlock5 is indeed the current development, but the Arduino support is far less than in version 3.4 .
Some just bring valuable extensions, like https://www.technologieservices.fr/mblock-extensions-ts-3-4-11-ress-175416.html

cmcknight, 2019-10-31:
> The latest source code is v3.4.5, there is no opensource code of 3.4.11 neither v5
>
> Anyway my question is only about if i can name my fork of mblock as "mblock_MOD" and make users download a mblock_MOD.exe compiled by me
>
> It's legal to do it or not?

Were I to do something like this I would approach it as follows:

1. Refactor the makeblock Arduino library from the current big ball of mud it is into something way more modular.
2. Consider using something like S4A or Blockly as the basis of the GUI.

This will avoid any copyright issues and will likely provide a more open platform given that mBlock 3.x requires proprietary tools/libraries for Scratch and mBlock 5 source doesn’t seem to be available.

Best regards,

Charles

bpau, 2019-11-01:
I don't know what exactly you want to improve om mBlock 3 Arduino support. It is better than the S4A , and the "technolgieservices" extensions make it even better, extending the orion.ino with new sensors. (it's open source as well). However the Arduino support in mBlock5 is letting me down. So I'm looking to alternatives like Ardublock and Ardublockly. Not as good as the interactive mode of mBlock 3 , but the Roberta-lab might improve ... https://lab.open-roberta.org/ . mBlock does not support the compilation of array's or lists in Arduino and Micro:bit, Strange ... others do.

cmcknight, 2019-11-01:
I would want to move away from rewriting Adobe Air. I write Arduino code in C/C++ personally but one of the biggest issues with building mBlock 3.x was that the version of Scratch that is based on uses Adobe Air. The graphical front end for block coding can, as you note, be done in any number of toolkits. I mentioned S4A because it was one of the tools I had looked at back with I was considering writing an IDE to replace mBlock a few years ago.