Makeblock-official/mBlock #92: Segmentation fault on Arch Linux

aa2006, 2021-09-14:
Running `./mblock` from the directory gives the following output:

```
/home/user/Downloads/mBlock/resources/tools/arduino
app listening on port 7070:
Segmentation fault (core dumped)
```

Nuclear-Squid, 2023-11-16:
Hey man, did you find a solution ? I’m running in the same problem. When running with valgrind I get this error, but I can’t find any documentation on the subject.

```
[71664:1116/105148:FATAL:content_main_runner.cc(653)] Check failed: base::allocator::IsAllocatorInitialized(). 
#0 0x000001e5801e <unknown>
#1 0x000001e6dd1b <unknown>
#2 0x000001221172 <unknown>
#3 0x00000122007a <unknown>
#4 0x000003479173 main
#5 0x000007ecbcd0 <unknown>
#6 0x000007ecbd8a <unknown>
#7 0x000000578cc9 <unknown>
```

adamtherookie, 2023-11-16:
No I haven't found a solution

hobbit-linux-42, 2023-12-01:
i use arch too and i have the same problem